bambulab/BambuStudio #4318: Support Interface not generating when using tree supports

TomDoesCars, 2024-06-18:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Ryzen 7 3700x, RTX4060Ti, 32gb DDR4

### Printer

P1S

### How to reproduce

Open a model, slice using tree supports.

### Actual results

The tree supports do not generate any kind of support interface, other than a line directly on top of the trees. No settings changes seem to impact this. Normal supports work as they have in all the previous releases I have used.
[image]
[image]
[image]

### Expected results

Support interface should generate.

### Project file & Debug log uploads

[Part.zip](https://github.com/user-attachments/files/15884911/Part.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

AJZein, 2024-06-18:
This happens with 'Tree Organic' support style only I've noticed but not sure if its intentional or not. The 'Default' option is defaulting to 'Tree Organic' in this case

OwainJ, 2024-06-20:
I'm seeing the same thing, since the last update, where `Tree(auto)` with `Default` / `Tree Organic` style doesn't generate the interfaces properly:
[image]
[image]
[image]

tangketan, 2024-06-25:
@OwainJ  set the "max bridge length" to 0 will fix your issue. We'll change the default behavior in the next version.

ttt888zzz, 2024-08-27:
Please try updating the latest version 1.9.4.
https://bambulab.com/en/download/studio

closed 2024-08-28 by ttt888zzz

Zabinator, 2025-03-03:
Hi I just found this thread. I am on the latest version and this still seems to be an issue.

Thilenios, 2025-03-06:
I can also confirm that I am not getting any support interfaces unless I manually paint support locations.

[image]