bambulab/BambuStudio #4321: Generic PCTG shows as PETG

andre1808, 2024-06-18:
### Bambu Studio Version

Most recent, also mobile app

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11, iOS 17

### Printer

P1S

### How to reproduce

Go to your AMS, select Generic PCTG and confirm. It will still show as PETG

### Actual results

[image]

[image]

### Expected results

It should say PCTG. 

### Project file & Debug log uploads

Project independent. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

andre1808, 2024-06-18:
Sorry about this very short description. I’m on my mobile. But I guess it’s obvious what the issue is.

SaltWei, 2024-06-26:
@QingZhangBambu Should be fixed， right?

QingZhangBambu, 2024-06-26:
@andre1808 it had been fixed. please check it.

andre1808, 2024-06-26:
Hi. Thanks for your support. Can you please explain how I can check that? There’s no firmware update available to my printer, neither the iOS app. I’ve just checked on the iOS app and the problem is still present.

QingZhangBambu, 2024-06-26:
@andre1808 hi, you need to restart your printer, the filaments will update automatically. Then the type will be right

andre1808, 2024-06-26:
@QingZhangBambu thanks for your response.
Maybe I wasn’t clear initially. I’m not using Bambulab PCTG.
To test this, I’ve done the following now:
- I’ve set the AMS spool to generic PCTG using the Bambu Lab iOS app (was still showing as PETG afterwards)
- turned off the printer, waited 5-10 seconds and turned it on again
- Checked within the app again and it still was showing as PETG. When I edit the spool, it showed as Generic PCTG. 
[image]

[image]

QingZhangBambu, 2024-06-27:
OK， I misunderstood. I thought you set filament on AMS， not by the app. I will talk to the concerned staff.