bambulab/BambuStudio #4322: Incorrect Filament Shown in Studio After Automatic Spool Switch

MadMax389, 2024-06-18:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Printer

P1P

### How to reproduce

I had set up slot A1 as a remnant spool (only 50g left on it). Slot A4 was a new spool set up as the same material and color as slot A1.

### Actual results

The printer automatically switched to slot A4 when slot A1 ran out, as expected, and is printing properly. However, the device screen (in Bambu Studio) is showing the filament being fed from the external spool, not slot A4. 
[image]

The Handy App shows the correct spool.
[image]

### Expected results

Indicate correct spool feeding after a spool switch.

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2024-06-21:
@MadMax389 
Hi. Can you provide logs for Studio and printer

MadMax389, 2024-06-21:
Studio and Printer log files [here](https://1drv.ms/f/s!AoYO2Eapmokqi5o3Huit3fJE2NG2NA?e=lr2L4h). I posted this issue while the printer was printing so the timestamp of the post is the approximate time the issue happened.

MadMax389, 2024-06-29:
Looks like the X1C has the same issue.
https://www.facebook.com/groups/bambulabofficialusers/posts/1214542722823078/