bambulab/BambuStudio #4323: Save custom filament profiles on printer/AMS itself

FloodedZebra0, 2024-06-19:
**Is your feature request related to a problem? Please describe.**

When on the bambu slicer, I would have to constantly select my custom filament profile before printing. As the unit itself has no way of storing custom profiles. I noticed this when I would select my custom profile. (Example, "Custom Nylon.") Attempt to sync to AMS/Printer, then trying to resync AMS profiles to slicer, this is where it would revert back to "Generic PA" instead of "Custom Nylon"

**Describe the solution you'd like**
(A clear and concise description of what you want to happen.)

Some way to sync custom filament profiles to the printer/AMS itself rather than manually selecting the respective profile before each and every print. I feel this Will make things more efficient as I don't have to constantly select it myself.

**Describe alternatives you've considered**
Simply selecting it manually before every print

**Additional context**

N/A

SaltWei, 2024-06-26:
https://github.com/bambulab/BambuStudio/assets/110660407/da8406a0-6d0a-4b3e-9717-070c25905edc

@FloodedZebra0 Seems to be working under my environment

SaltWei, 2024-06-26:
You need to create a custom filament, rather than editing a existed filament and renaming to be new name.
Creating a new custom filament will create new filament id which is used to match when sync.
But renaming is fake operation and the filament id is still the old existed one.

FloodedZebra0, 2024-06-26:
Thank you, did not notice that when browsing around.

closed 2024-06-26 by FloodedZebra0