bambulab/BambuStudio #4325: Mac version after save 3mf cannot open it again.

long187, 2024-06-19:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS Big Sur

### Additional system information

[image]

### Printer

P1S

### How to reproduce

open an existing 3mf file, save as , use an other window open it

### Actual results

null

### Expected results

null

### Project file & Debug log uploads

null

### Checklist of files to include

- [ ] Log file
- [X] Project file

ttt888zzz, 2024-06-20:
Could you please upload the source file

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

closed 2025-04-16 by Haidiye00