bambulab/BambuStudio #4328: Option for 0 purge on full-layer  colour change

mctazman, 2024-06-19:
**Is your feature request related to a problem? Please describe.**
Ever since the AMS was released, people have been trying to minimize the fila-poop. I've successfully tested a 0 purge colour change, ergo virtually no waste.

**Describe the solution you'd like**
An option to, when there's a full-layer colour change, choose 0 purge value. The reason I'd like this feature is because people will either forget to set the value to 0, or forget to change it back to normal afterwards, causing unexpected results.

**Additional context**
Video of the succesfull test: https://www.youtube.com/watch?v=xnTfDI3kfK4
The set purge values:
[image]
The used STL:
[Magnificent Migelo-Robo.stl.zip](https://github.com/user-attachments/files/15899583/Magnificent.Migelo-Robo.stl.zip)
The results:
[image]
[image]
[image]

I added manual colour changes on layers 33 & 67, 0.2mm layer height, 3 bottom/5 top layers

GuzziRaz, 2024-06-19:
I also tried that, in a subtly different way - I just set the flush multiplier to zero. That too worked like a champ (for certain models, such as your sample).

What exactly is your suggestion though? You seem to mean just an "on/off" check box for zero purge. In what context? Perhaps in the Send Print Job dialog (i.e. along with the options for flow calibration, bed levelling etc.)?

Even with such option implemented, the per-filament example is more fine grained than what such global on/off option could achieve so it would still have merits, eg. when you only want a few filaments set to zero flush but keep flushing other ones. So perhaps there could also be a warning pop-up such as this before printing:

  Warning: One or more flushing volumes are set to zero

    [OK]   [Cancel]   [Don't warn again for this project]

In fact, personally I'd be fully satisified with that alone. I see less need for a global check box.

mctazman, 2024-06-21:
That would be perfect, and a LOT better than my idea. :D