bambulab/BambuStudio #4329: FR: Fix model for Linux

tim-gromeyer, 2024-06-19:
**Is your feature request related to a problem? Please describe.**
Hello, I would like to use the "Fix model" option on Linux.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
I saw, that this functionality is implemented using the windows sdk (`src/slic3r/Utils/FixModelByWin10.cpp`)

alanLacey44, 2024-06-20:
Yes, me too.  Don't like having to go to Windows for non-manifold fixes.  Sadly I can't install the studio exe under Wine in my Mint 21.1

freckletonj, 2024-08-29:
same, this would be great!

louhy, 2024-10-20:
"Use Windows" is a bit of an infuriating suggestion. I'm not installing another OS just for this, and certainly not paying for it either. We need this on Linux. Or at least have it tell us where the problem is!

louhy, 2024-10-20:
Why is this OS specific, anyway? What's it doing that isn't portable?

tim-gromeyer, 2024-10-21:
It's using the Windows API to repair the model.

louhy, 2024-11-01:
...something.. built into Windows? _Weird_ (no wonder it's so bloated :stuck_out_tongue_closed_eyes:)