bambulab/BambuStudio #4331: load mtl in obj:failed to parse / The file does not contain any geometry data

brainysi, 2024-06-19:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 10

### Additional system information

Ryzen5 3600x, 32 gb ram, rx 6800 16 gb

### Printer

Bambulab A1 mini

### How to reproduce

1. Make model with multiple subobjects different colors on tinkercad
2. Export as obj
3. Unzip the zip file provided
4. Drag and drop the obj into bambu studio.

### Actual results

[image]

### Expected results

It should just work, as it did on the previous version

### Project file & Debug log uploads

Cannot upload the project file as nothing is imported.
Uploading a test model with which to reproduce the problem.
[Earing 1.zip](https://github.com/user-attachments/files/15904970/Earing.1.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-06-20:
[image]
I have used your OBJ file and it can be displayed normally. Is your software the latest？

brainysi, 2024-06-20:
Retested now on a different pc and it works for me too. Will retest at home in the evening.

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

Gh0stBustr, 2025-08-07:
Same for me!

Haidiye00, 2025-08-07:
hello,please provide your files

Gh0stBustr, 2025-08-07:
Turned out to be a lack of permissions on the file in macOS.
Bambu Studio should provide a more meaningful error message in this case.