bambulab/BambuStudio #4332: Turn off fan and increase nozzle temperature when ironing

vadixidav, 2024-06-19:
**Is your feature request related to a problem? Please describe.**
The tops of uneven surfaces (due to sagging bridging or otherwise) have difficulty being ironed and leave underextrusion lines in many cases due to the rolling of the ironing filament over when it is too cool. Increasing the temperature of the nozzle prior to ironing is so incredibly important for the success of the ironing. However, having the nozzle temperature so high during printing generally can cause stringing and other failures. Here is an example of a situation where I increased the temperature mid-ironing:

[image]

This was printed with an ironing speed of 50mm/s, a bit too high. The left tile is ironed at 205C and the right tile is ironed at 230C. The difference is clear. Notice how the left tile has lots of underextrusion patches where the nozzle rolls/balls up the filament because it is too low of a temperature and doesn't adhere and spread out onto the underlying surface and instead gums up and gets stuck to existing rolls, continuing the pattern, until eventually it breaks and then repeats. I have printed around 50 various prints to dial in these under-extrusion lines. Here are some more interesting ones:

[image]

Notice how the underextrusion lines (or rolls, as I call them) happen in distinct locations where there is sagging in the bridging. The best of these were printed with an ironing speed of 10-20mm/s. See below for the bridging locations:

[image]

If the surface is totally flat, it seems to have no issue, but when the surface sags the filament has a hard time adhering to the layer underneath, even if it is only ten microns of a difference.

However, it is clear that these lines all go away when I turn the temperature up from 205C (normal printing, to prevent stringing and improve bridging to prevent this issue) to 230C (only when ironing, which significantly improves if not removes this issue).

**Describe the solution you'd like**

[image]

In the cooling menu (above) in the filament settings, please add a section called **Ironing**. This section should contain the following options:

* [ ] Lower fan speed during ironing
* Ironing layer time fan speed (This is a percentage fan speed. Set this to 0% and the fan will turn off even if "Keep fan always on" is set.)
* Ironing layer time threshold (The fan will only be reduced in speed if the ironing will take longer than this many seconds. Set this to 0 to always lower the fan during ironing.)
* [ ] Wait before ironing for fan to slow down (This will wait until the fan is slowed down before beginning the ironing process)
* [ ] Wait after ironing for the fan to speed up (This will wait until the fan is sped back up to normal before resuming printing after ironing)

[image]

In the filament menu (above) in the filament settings, please add a section called **Ironing**. This section should contain the following options:

* Ironing nozzle temperature (Degrees C temperature to use during ironing)
* Ironing layer time threshold (Only lower the nozzle temperature if the ironing will take longer than this many seconds. Set this to 0 to always lower the nozzle temperature during ironing.)
* [x] Wait before ironing for nozzle to warm up (This will wait until the nozzle is warmed up before beginning the ironing process)
* [x] Wait after ironing for the nozzle to cool down (This will wait until the nozzle is cooled down to normal before resuming printing after ironing)

**Describe alternatives you've considered**

It may be possible to print the entire print at a higher temperature currently, but this results in significant stringing and inferior bridges. This issue cannot be avoided by improving the bridges (I have extensively tested optimal bridging to resolve this issue) and it is also not resolved by going at a slower ironing speed than 20mm/s (such speeds yield no improvements and the rolling still happens).

One other alternative would be that it may be possible to implement a very similar thing for the nozzle temperature and fan speed before starting a long bridging session (my parts have bridging that goes on for about an hour all on the same layer). In this case the bridging may be improved further as they can print at temperatures lower and fan speeds higher than the rest of the print. However, I don't know if this fixes the crux of the issue, which is that ironing seems very inconsistent unless you raise the temperature during ironing. I would like for my surfaces to be totally flat and flawless.

**Additional context**

I am currently printing parts for organizers, but I may also want to print parts for bioreactors and other things, and making those watertight will be important. I think ironing will be key to ensuring all the layers are totally crystalline and free of bubbles in that case. I have not done extensive testing with PETG. All of the images you see above were done with ELEGOO PLA+ filament. The severity of this issue does change with the kind of filament you are using, but it seems especially bad with PLA filament. It is much better with ELEGOO PLA PRO filament, for instance, but it still occurs depending on the sagging of bridging, supports, and overhangs.

BlueByLiquid, 2024-08-18:
Second this. I need this feature as well

caizixian, 2024-10-19:
On eSUN PLA+, I found ironing with 0% fan (middle section of the photo) gives the best result. 
[image]