bambulab/BambuStudio #4333: Crash without Crash Detection

Armaggeden, 2024-06-20:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: 13th Gen Intel(R) Core(TM) i9-13900KF   3.00 GHz
128 GB RAM
Nvidia RTX 4090

### Printer

Bambu Lab X1 Series

### How to reproduce

Click Slice Buttin. Slice gets to about 15% and the program exits.

### Actual results

The Program exits or crashes. Normally there is some kind of crash report. I don't know if this program has one but it just exits. 

### Expected results

It should slice my plate so I can print. 

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15916988/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2024-06-24:
[#3318](https://github.com/bambulab/BambuStudio/issues/3318)
Hi, could you try these methods? sometimes the CPU may not be stable enough to support multithreading.

QingZhangBambu, 2024-06-24:
and could upload 3mf ?