bambulab/BambuStudio #4334: Printer and Process Selection Drop Downs appearing on wrong display

ohmdelta, 2024-06-20:
### Bambu Studio Version

01.09.01.66

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04

### Printer

P1

### How to reproduce

1. connect multiple monitors up
2. open bambustudio, create new project etc
3. click drop down menus for printer selection or process selection

### Actual results

selection menus show up on the wrong display - not on the bambustudio app

### Expected results

selection menus show should be under the current drop down tab

### Project file & Debug log uploads

na

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2024-06-24:
This is a known issue, and a flatpak version can currently be used instead.
https://flathub.org/apps/com.bambulab.BambuStudio

MackBambu, 2024-07-11:
Can you provide some screenshots and log files?