bambulab/BambuStudio #4339: Crashing on Macos Immediately after sending a print to my A1

Phil-Rader, 2024-06-21:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.5

### Additional system information

2.7 GHz Dual-Core Intel Core i5
Intel Iris Graphics 6100 1536 MB
8 GB 1867 MHz DDR3

### Printer

Bambu Lab A1

### How to reproduce

Sending a print to the A1 causes Bambu Studio to crash

### Actual results

Bambu Studio crashes

### Expected results

Application should stay open

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15922750/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

walterwongbbl, 2024-06-21:
Hi.  @Phil-Rader 
Can you provide a crash repoer.

lbreggi, 2024-06-29:
I have the same issue. just sent a ticket with all crash reports

walterwongbbl, 2024-09-04:
@Phil-Rader @lbreggi 
Hi. Are you still experiencing crashes recently

lbreggi, 2024-09-04:
Not after the latest update so far