bambulab/BambuStudio #4341: 支撑页面的中文描述错误 (与实际情况相反)

imzhuli, 2024-06-21:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

1. 点开 "准备" 页面
2. 选择"支撑"子页面
3. 将鼠标悬浮在 选项 "仅在打印板生成", 上. 
4. 可见描述错误

### Actual results

当选择被勾选时. 描述中是"不在模型表面生成, 只在热床上生成", 实际的切片是"仅在模型表面上生成支撑(on plate only), 而非在热床上生成"

### Expected results

应当将描述改为: "仅在模型表面上生成支撑(on plate only), 而非在热床上生成",

### Project file & Debug log uploads

无

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tangketan, 2024-06-26:
当勾选这个选项时，确实只从打印板生成支撑，不会从模型表面生成。这个表达没问题啊。

imzhuli, 2024-06-29:
[image]
我这两天打印别的模型时发现.. 这个问题偋不是文字问题. 而是选择"树状"时. 这个选项好像是没有意义的? (如图)