bambulab/BambuStudio #4342: Bambu Studio Flathub - Go Live

immortel32, 2024-06-21:
### Bambu Studio Version

1.9.2

### Where is the application from?

Built from source by yourself or others

### OS version

Linux mint 21.3

### Printer

Bambu Lab X1c

### How to reproduce

Install Bambu Studio with Flathub
Enable the Go Live option
Configure OBS Studio to read the file /home/myuser/.var/app/com.bambulab.BambuStudio/config/BambuStudio/cameratools/ffmpeg.sdp

### Actual results

I got a black screen (nothing happens)

### Expected results

I'm expecting to see the live camera on my printer.
If I do it with AppImage and use the file in /home/myuser/.config/BambuStudio/cameratools/ffmpeg.sdp, it is working fine.

The reason of using Flathub intead of AppImage is because it manages Bambu Studio upgrades. The last I check with AppImage it did not warn me when a new version was available.

### Project file & Debug log uploads

Can be done with any STL

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2024-06-24:
It is possible that the ffmpeg process failed to start. Please help check if there is an ffmpeg process. Bing provides studio logs to analyze the reason for the failure.

immortel32, 2024-06-25:
> It is possible that the ffmpeg process failed to start. Please help check if there is an ffmpeg process. Bing provides studio logs to analyze the reason for the failure.
This is what I get with the application from flathub
```
ps -ef | grep ffm
myuser        7284    7037  1 21:23 ?        00:00:00 [ffmpeg] <defunct>

```
This is what I get with the appimage
`
ps -ef | grep ffm
myuser        5773       1  1 21:20 pts/0    00:00:02 /home/myuser/.config/BambuStudio/cameratools/ffmpeg -fflags nobuffer -flags low_delay -analyzeduration 10 -probesize 3200 -f h264 -i pipe: -vcodec copy -f rtp rtp://127.0.0.1:1234
`

There is a lot of "sensitive data" in the logs. What do you need? (There is nothing in the logs with ffmpeg).

bambu123, 2024-06-25:
> ```
> ps -ef | grep ffm
> myuser        7284    7037  1 21:23 ?        00:00:00 [ffmpeg] <defunct>
> ```
>
> This is what I get with the appimage `ps -ef | grep ffm myuser 5773 1 1 21:20 pts/0 00:00:02 [...]

I need the log lines with "MediaPlayCtrl".

immortel32, 2024-06-25:
Hello,
Here's the output for the command grep -a MediaPlayCtrl *
```
debug_Mon_Jun_24_21_06_07_4.log.0:2024-06-24 21:06:16.319814[Thread 0x00007fb6a6ad1bc0]:MediaPlayCtrl switch machine: 00M00A2C0604476
debug_Mon_Jun_24_21_06_07_4.log.0:2024-06-24 21:06:16.319840[Thread 0x00007fb6a6ad1bc0]:MediaPlayCtrl::SetStatus:  tutk_state: enable
debug_Mon_Jun_24_21_06_07_4.log.0:2024-06-24 21:06:20.843458[Thread 0x00007fb6a6ad1bc0]:MediaPlayCtrl::TogglePlay
debug_Mon_Jun_24_21_06_07_4.log.0:2024-06-24 21:06:20.845908[Thread 0x00007fb6a6ad1bc0]:MediaPlayCtrl::SetStatus: Initializing... tutk_state: enable
debug_Mon_Jun_24_21_06_07_4.log.0:2024-06-24 21:06:20.848256[Thread 0x00007fb6a6ad1bc0]:MediaPlayCtrl: bambu:///tutk?uid=E3LH************111A&authkey=********&passwd=******&region=us&device=00M00A2C0604476&net_ver=01.09.03.01&dev_ver=01.08.00.00&cli_id=128eac7f-ad93-4074-9221-9c5aed70741b&cli_ver=01.09.03.50
debug_Mon_Jun_24_21_06_07_4.log.0:2024-06-24 21:06:20.877576[Thread 0x00007fb6a6ad1bc0]:MediaPlayCtrl::SetStatus: Loading... tutk_state: enable
debug_Mon_Jun_24_21_06_07_4.log.0:2024-06-24 21:06:20.878111[Thread 0x00007fb6175fb6c0]:MediaPlayCtrl: start load
debug_Mon_Jun_24_21_06_07_4.log.0:2024-06-24 21:06:22.115914[Thread 0x00007fb6175fb6c0]:MediaPlayCtrl: end load
debug_Mon_Jun_24_21_06_07_4.log.0:2024-06-24 21:06:22.115973[Thread 0x00007fb6a6ad1bc0]:MediaPlayCtrl::onStateChanged: 0, last_state: 5
debug_Mon_Jun_24_21_06_07_4.log.0:2024-06-24 21:06:22.115986[Thread 0x00007fb6a6ad1bc0]:MediaPlayCtrl::onStateChanged: size: 1280x720
debug_Mon_Jun_24_21_06_07_4.log.0:2024-06-24 21:06:22.115994[Thread 0x00007fb6a6ad1bc0]:MediaPlayCtrl::SetStatus: Playing... tutk_state: enable
debug_Mon_Jun_24_21_06_07_4.log.0:2024-06-24 21:06:22.116647[Thread 0x00007fb6a6ad1bc0]:MediaPlayCtrl::onStateChanged: 0, last_state: 0
debug_Mon_Jun_24_21_06_07_4.log.0:2024-06-24 21:06:22.116655[Thread 0x00007fb6a6ad1bc0]:MediaPlayCtrl::onStateChanged: skip when task not finished
debug_Mon_Jun_24_21_06_07_4.log.0:2024-06-24 21:06:22.117927[Thread 0x00007fb6a6ad1bc0]:MediaPlayCtrl::onStateChanged: 0, last_state: 0
debug_Mon_Jun_24_21_06_07_4.log.0:2024-06-24 21:06:22.659493[Thread 0x00007fb6a6ad1bc0]:MediaPlayCtrl::onStateChanged: 2, last_state: 0
debug_Mon_Jun_24_21_06_07_4.log.0:2024-06-24 21:06:26.025111[Thread 0x00007fb6a6ad1bc0]:MediaPlayCtrl::ToggleStream: bambu:///tutk?uid=E3LH************111A&authkey=********&passwd=******&region=us&device=00M00A2C0604476&net_ver=01.09.03.01&dev_ver=01.08.00.00&cli_id=128eac7f-ad93-4074-9221-9c5aed70741b&cli_ver=01.09.03.50
debug_Mon_Jun_24_21_06_07_4.log.0:2024-06-24 21:09:24.534408[Thread 0x00007fb6a6ad1bc0]:MediaPlayCtrl::ToggleStream: bambu:///tutk?uid=E3LH************111A&authkey=********&passwd=******&region=us&device=00M00A2C0604476&net_ver=01.09.03.01&dev_ver=01.08.00.00&cli_id=128eac7f-ad93-4074-9221-9c5aed70741b&cli_ver=01.09.03.50
debug_Mon_Jun_24_21_06_07_4.log.0:2024-06-24 21:09:26.071458[Thread 0x00007fb6a6ad1bc0]:MediaPlayCtrl::ToggleStream: bambu:///tutk?uid=E3LH************111A&authkey=********&passwd=******&region=us&device=00M00A2C0604476&net_ver=01.09.03.01&dev_ver=01.08.00.00&cli_id=128eac7f-ad93-4074-9221-9c5aed70741b&cli_ver=01.09.03.50
debug_Mon_Jun_24_21_15_48_4.log.0:2024-06-24 21:15:54.150820[Thread 0x00007f074ffacbc0]:MediaPlayCtrl switch machine: 00M00A2C0604476
debug_Mon_Jun_24_21_15_48_4.log.0:2024-06-24 21:15:54.151234[Thread 0x00007f074ffacbc0]:MediaPlayCtrl::SetStatus:  tutk_state: enable
debug_Mon_Jun_24_21_23_29_4.log.0:2024-06-24 21:23:36.072029[Thread 0x00007f6326f3cbc0]:MediaPlayCtrl switch machine: 00M00A2C0604476
debug_Mon_Jun_24_21_23_29_4.log.0:2024-06-24 21:23:36.087362[Thread 0x00007f6326f3cbc0]:MediaPlayCtrl::SetStatus:  tutk_state: enable
debug_Mon_Jun_24_21_23_29_4.log.0:2024-06-24 21:23:42.931775[Thread 0x00007f6326f3cbc0]:MediaPlayCtrl::ToggleStream: bambu:///tutk?uid=E3LH************111A&authkey=********&passwd=******&region=us&device=00M00A2C0604476&net_ver=01.09.03.01&dev_ver=01.08.00.00&cli_id=128eac7f-ad93-4074-9221-9c5aed70741b&cli_ver=01.09.03.50
```

posita, 2024-09-04:
I am also experiencing this issue. I get about five seconds of video, then it freezes, and I also have a `ffmpeg [defunct]` process that hangs around (i.e., is not being reaped by the parent process) until I quit BambuStudio. If I stop-and-start the video from within BambuStudio, I get another five seconds (and so on).

posita, 2024-09-04:
FYI, I would not need this fixed if #1536 was addressed.