bambulab/BambuStudio #4343: Crash when selecting plate 3 after slicing

StoneCut, 2024-06-21:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 14.5 (23F79)

### Additional system information

Apple M1 Max, 32 GB Ram

### Printer

Bambu Lab X1C

### How to reproduce

1. Slice the first (Designer's) print profile of the following model (Plate 1):
https://makerworld.com/en/models/46520#profileId-48304 - Cannot attach, too large
2. Select plate 3.
3. Bambu Studio crashes

### Actual results

Bambu Studio crashes

### Expected results

Bambu Studio switches to Plate and doesn't crash,

### Project file & Debug log uploads

[bambustudio_crash.txt](https://github.com/user-attachments/files/15927876/bambustudio_crash.txt)

### Checklist of files to include

- [X] Log file
- [ ] Project file

QingZhangBambu, 2024-06-24:
https://github.com/bambulab/BambuStudio/issues/3318
Hi, could you try these methods? sometimes the CPU may not be stable enough to support multithreading.

StoneCut, 2024-06-24:
I believe above methods are not applicable as this is a Mac.