bambulab/BambuStudio #4344: Dowels goes missing after fixing model

jchau57, 2024-06-21:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Home

### Additional system information

[image]
[image]

### Printer

Bambu Labs P1S

### How to reproduce

1. Imported the model below
2. Created a cut with dowel connectors along vertical centerline of the model
3. Model is cut with the dowel holes but there are non-manifold edge errors (see image 1, 14 non-manifold edges errors in bottom right)
4. If I fix/repair the model, the errors go away but the model has shifted and the dowel holes do not line up anymore (see image 2)
5. If I export the model as a STL file the dowel holes are not shown at all (see image 3)
6. I am unable to do a boolean mesh and there is no option to assemble the holes to the model

Here is a link to the model I am working with:
https://www.printables.com/model/148007-star-trek-bird-of-prey/files

### Actual results

See above

### Expected results

I created the dowel holes along the vertical centerline of the model.  I expect the dowel holes to stay there when fixing the model.  I also expect the dowel holes to show up when I export the model to STL

### Project file & Debug log uploads

[bird_of_prey.zip](https://github.com/user-attachments/files/15930665/bird_of_prey.zip)
[debug_Fri_Jun_21_09_52_41_2584.log.zip](https://github.com/user-attachments/files/15930680/debug_Fri_Jun_21_09_52_41_2584.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

jchau57, 2024-06-22:
[bird_of_prey.zip](https://github.com/user-attachments/files/15941982/bird_of_prey.zip)

jchau57, 2024-06-22:
Please see updated bird_of_prey.zip file.

Haidiye00, 2024-06-24:
Hello, can you provide the original stl model？thanks

jchau57, 2024-06-25:
[Uploading bird_of_prey original.zip…]()
Here's the original stl model.

jchau57, 2024-06-25:
[Uploading bird_of_prey original.zip…]()
Hello, here is the original stl file of the model.

jchau57, 2024-06-25:
[bird_of_prey original.zip](https://github.com/user-attachments/files/15965314/bird_of_prey.original.zip)

jchau57, 2024-06-25:
Hello, here is the original stl model file.

Haidiye00, 2024-06-27:
[image]
Hello, I did not reproduce your question. Can you record the operation process of a GIF? thank you

jchau57, 2024-06-28:
Ok, I reproduced my problem step by step.
1) I created a vertical cut with 2 dowel connectors.  Attached is the model file and a screenshot after the cut.  As you can see the front portion of the model has 12 non-manifold edges error.
[image]
[bird_of_prey 01.zip](https://github.com/user-attachments/files/16023366/bird_of_prey.01.zip)

jchau57, 2024-06-28:
2. Here a screenshot and model file after I perform a repair of the non-manifold edges.  As you can see the dowel holes are not lined up anymore.
[image]
[bird_of_prey 01A.zip](https://github.com/user-attachments/files/16023396/bird_of_prey.01A.zip)

jchau57, 2024-06-28:
3) Since I could not get the dowel holes to line up, the other thing I tried was to export it to STL before repairing the non-manifold edges.  However, the result is the dowel holes are missing.  Here is the screenshot and STL file.
[image]
[bird_of_prey 02.zip](https://github.com/user-attachments/files/16023421/bird_of_prey.02.zip)

jchau57, 2024-08-04:
Is there any update to this reported problem?  I've provided as much detail as I can already.  It's over 1 month.

Haidiye00, 2024-10-14:
[image]
I noticed that after cutting, an object was suspended in the air. How was this done?

jchau57, 2024-10-14:
I don't know.  That was the result after I cut the model.

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00