bambulab/BambuStudio #4347: Re-Add the possibility to disable vibration calibration as in the official video

Roemer, 2024-06-21:
**Is your feature request related to a problem? Please describe.**
In this video (https://wiki.bambulab.com/en/x1/manual/print-from-bambu-studio) you can see that there is a checkbox to enable or disable "vibration calibration" but I do not see that. But it could be very handy to have that to if I want to print multiple smaller things one after another.

**Describe the solution you'd like**
Add the checkbox as in the video

**Describe alternatives you've considered**
Manually modifying the G-Code. Works probably but is tedious.

poudenes, 2024-11-09:
hmm... I don't see this option when I want print to my A1 Mini....  Bambu Studio 1.9.7.52

gschintgen, 2024-12-01:
Yes, that vibrating at the start of each print is quite annoying. I don't move my printer between prints so I don't see how the results could change from print to print. It just seems superfluous.

gschintgen, 2024-12-01:
Hm, apparently this short vibration routine at the start of each print is necessary: https://github.com/bambulab/BambuStudio/issues/1424#issuecomment-1477173031

NeariX67, 2024-12-03:
I don't think it's necessary. If i have many smaller prints, it just a matter of time until a neighbor knocks at the door for something totally unnecessary. I'm fine with doing it once after startup, but not before each print.