bambulab/BambuStudio #4348: A1 Extruder Rattle Noise at the beginning of each print

Roemer, 2024-06-21:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab A1

### How to reproduce

1. Slice any model in Bambu Lap
2. Print it

### Actual results

In the very start phase, the extruder unit moves to the center and then the extruder rattles about 4 times as discussed in this reddit post with video: https://www.reddit.com/r/BambuLab/comments/1cmuef2/comment/l9cu95u/
I guess there is some faulty G-Code somewhere in the start code.

### Expected results

There should be no rattling / trying to extrude or retract at that phase of the print as the nozzle is not yet heated up.

### Project file & Debug log uploads

Not needed, this is with every project and for multiple persons (see reddit).

### Checklist of files to include

- [X] Log file
- [X] Project file

gaolegao-lx, 2024-06-25:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

Flexigem, 2024-07-19:
@Roemer 
I updated this thread with an answer from Bambu:
https://github.com/bambulab/BambuStudio/issues/3426