bambulab/BambuStudio #4350: BambuStudio will not work on latest official Ubuntu 24.04 LTS, broken dependencies of AppImage

gfornax, 2024-06-22:
### Bambu Studio Version

Bambu_Studio_linux_ubuntu-v01.09.03.50

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 24.04 LTS

### Additional system information

cat /etc/debian_version 
trixie/sid
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04 LTS
Release:	24.04
Codename:	noble

### Printer

N/A

### How to reproduce

starting via ./Bambu_Studio_linux_ubuntu-v01.09.03.50.AppImage

Problem occured after upgrade to latest Ubuntu LTS version. Before it worked fine. Library dependency issue.

### Actual results

./Bambu_Studio_linux_ubuntu-v01.09.03.50.AppImage         
/tmp/.mount_Bambu_nnAiff/bin/bambu-studio: error while loading shared libraries: libjavascriptcoregtk-4.0.so.18: cannot open shared object file: No such file or directory

And a lot of other libraries if I try to fix them manually

### Expected results

start of application

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Andrew-Pynch, 2024-06-22:
@gfornax +1 for this issue, I'm also unable to build from source for similar reasons

MackBambu, 2024-06-24:
https://github.com/bambulab/BambuStudio/blob/master/BuildLinux.sh#L111
There is an interim solution

MackBambu, 2024-06-24:
This is a known issue, and flatpak version can currently be used instead
https://flathub.org/apps/com.bambulab.BambuStudio

closed 2024-06-25 by MackBambu

FlizzerMDX, 2025-08-21:
For me flatpak version always crash, someones help ?