bambulab/BambuStudio #4354: Add an optional raft interface ironing feature.

vitaliecartera, 2024-06-22:
**Is your feature request related to a problem? Please describe.**
The print quality of the supported surface is always visibly rougher than the rest of the print.

**Describe the solution you'd like**
Apply an ironing step on the topmost raft interface layer, before laying down the first supported (clean) layer. This, could potentially significantly improve the quality of the supported surface, since the fondation of the raft interface would theoretically be  smoother.

**Describe alternatives you've considered**
This feature could be added first as an expiremental one. Would probably make sense if this feature is enabled/present only when there are more than one interface layers configured, and/or when the raft interface material is different (special) than the model’s. Although, ironing between the layers of the same type of material might in some cases improve the support separation as well.

**Additional context**
Note: this idea was NOT tested by the author of this request. However, [some users report](https://youtu.be/xs2urfM0MRM?si=EghZYXVBJUKii4qv) successful separation of the stacked prints, when the lower item has the ironing applied on its topmost layers. So, theoretically, this could improve the support separation and the overal quality of the supported surface (with the proper tweaking). TBD

spacepost, 2025-03-06:
I would like to see this as well for both rafts and support top layers for enabling smooth bottom finishes when using different support materials such as HIPS as support for ABS with 0mm interface spacing.

edzme, 2025-08-24:
this is a great idea, seems like getting that top surface of the support as smooth as possible (like the plate) might just work to achieve clean smooth bottom surfaces of the print especially for large flat bottoms

astral-alpaca, 2026-01-28:
According to https://wiki.bambulab.com/en/software/bambu-studio/release/release-note-2-5-0 this feature is already implemented. Can this be closed?