bambulab/BambuStudio #4355: Windows APP and printer not in sync

somedesa, 2024-06-22:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

win 11

### Additional system information

Gamer system, 64 gigs of ram, 129k cpu, 3080-ti win 11

### Printer

p1p

### How to reproduce

no idea

### Actual results

Win Software shows lane three of P1P/AMS in use.  P1P is using lane 4.  Layer 1 was the correct lane (3), I left and came back and it switched to AMS lane 4.  I shut down the pc software and restarted and went to device tab and still shows lane 3 printing on the device (only one printer here) while on the actual device  (P1P) lane 4 is in use.

### Expected results

correct lane shown and in use? printer firmware updated yesterday

### Project file & Debug log uploads

[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

somedesa, 2024-06-22:
as second image shows, pc app thinks lane 4 is empty while is in use on P1P

somedesa, 2024-06-22:
[organizer slim.zip](https://github.com/user-attachments/files/15937564/organizer.slim.zip)
project file

walterwongbbl, 2024-06-28:
Hi  @somedesa 
Can the Studio recover normally after the printer is restarted? If the display is still in error, can you provide the Studio's LOG file?

closed 2024-08-08 by walterwongbbl