bambulab/BambuStudio #4356: PA profiles are lost at printer after power outage

longtimeago, 2024-06-22:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04.4

### Printer

Bambulab X1C Carbon Combo

### How to reproduce

1. Perform Flow Dynamics calibration for a filament
2. Start printing any model from that filament using created PA profile
3. Turn printer off simulating power outage
4. Turn printer on
5. Resume printing

### Actual results

PA profile for the filament becomes “none” and there is NO way to adjust it without canceling the print job.
As a result, observe corrupted print.

For some filaments and models the result of resuming the job may be acceptable but most of the time the printed item is trashed.

### Expected results

I’d expect 2 ways of fixing the issue:

Give a way to adjust PA profile after power outage before resuming the print
Fix the Firmware/Studio in a way that PA profiles are not lost after power outage

### Project file & Debug log uploads

No logs or project file, sorry

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2024-06-24:
Could you reproduce the problem and share the Studio logs and printer logs?

longtimeago, 2024-06-26:
@zhimin-zeng-bambulab attached studio log files to the description. How could I attach printer log file? It's of 488 Mb size for unknown reason. I didn't include photos from camera or g-code when was exporting

zhimin-zeng-bambulab, 2024-08-06:
https://wiki.bambulab.com/en/x1/troubleshooting/upload-log

Could you reproduce the problem again, and then upload the machine logs according to the wiki's instructions?

gaolegao-lx, 2024-08-07:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

longtimeago, 2024-08-19:
@zhimin-zeng-bambulab @gaolegao-lx here it is `US240819833001`, sorry for the delay