bambulab/BambuStudio #4357: Dropdowns do not work properly with "x-mouse" style "window tracking" enabled

swinokur, 2024-06-22:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 7

### Printer

X1C

### How to reproduce

1. Enable "window tracking" in Windows 7 (or later?). This means that the window focus will automatically move to whatever window the mouse is in, without clicking. You can find a piece of software to easily manipulate this feature here: https://joelpurra.com/projects/X-Mouse_Controls/
2. with "activate window tracking" ticked and delay set to 0 ms, attempt to use the dropdowns in Bambu Studio. For example, try to change filament types, select a different printer, etc in the "prepare" tab.
3. You will notice that the dropdowns "vanish" before the mouse can move to select them

### Actual results

As mentioned, the dropdowns vanish before they can be selected by mouse.

The current workaround is to remember to run the above mentioned "x-mouse" controls application and disable window tracking when using Bambu Studio.

### Expected results

Ideally the dropdowns are not interpreted by windows as separate "windows" but actually as dropdowns within the application so that selection can work properly inside the application. 

### Project file & Debug log uploads

I don't think there are any log files needed for this UI issue, but if any can help please let me know which log(s) to enable (if any) for tracking.

### Checklist of files to include

- [X] Log file
- [X] Project file

Noodle-Bug, 2024-08-05:
This happens on Linux with Hyprland as well, rendering all dropdowns unusable.

zeffo, 2024-08-11:
Can reproduce on hyprland too, changing the `follow_mouse` setting is a temporary solution

Noodle-Bug, 2024-08-14:
> Can reproduce on hyprland too, changing the `follow_mouse` setting is a temporary solution

I can't seem to get it to work even with changing `follow_mouse`. So far my only workaround has been to switch over to Plasma and use it there, which is unfortunately kind of a major stopper when I'm in the middle of other things.

bambu123, 2024-08-28:
This is a known problem, but we haven't found a solution yet.

zeffo, 2024-09-09:
Hi, this seems to be fixed on BambuStudio-01.09.05.51.

swinokur, 2024-09-09:
> Hi, this seems to be fixed on BambuStudio-01.09.05.51.

It is still not fixed on that version for me.

gw0, 2024-12-30:
It seems I hit the same problem with Bambu Studio 1.10.1.50 (Ubuntu 20.04 release) in Qubes OS 4.2.3 in `debian-12-xfce` qube both using Xfce4 (also in `debian-10` qube using GNome 3.38). Dropdown menus are unusable, they show up for a split second then disappear. Same behavior for short or long click, right click, or scrolling. Because the UI is not designed for mouse-less usage it is also impossible to cycle through dropdown options with arrow keys.

Xfce4 options such as disabling "Automatically give focus to newly created windows", "Automatically raise windows when they receive focus", "Raise window when clicking inside application window" "Activate focus stealing prevention", "Honor standard ICCCM focus hint", and "When a window raises itself: bring to current workspace" have no effect.