bambulab/BambuStudio #4359: Right click frequently does not bring up right click menu

ThatSlacker, 2024-06-22:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

- Load a model
- Right click the model

### Actual results

Right click menu appears some of the time.  Other times it's interpreted as a panning move and the model shifts very slightly

### Expected results

Right click menu appears.  Panning movements that are an extremely brief duration (e.g. 100ms or less and less than a few pixels movement) are suppressed

### Project file & Debug log uploads

[debug_network_Sat_Jun_22_15_11_11.log.enc.zip](https://github.com/user-attachments/files/15939512/debug_network_Sat_Jun_22_15_11_11.log.enc.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-06-25:
Can you record a GIF? I haven't reproduced it yet

ThatSlacker, 2024-06-26:
Absolutely.  In this you can see the first right click (red dot indicates right click) moves it slightly and the second right click brings up the menu.
[image]

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00