bambulab/BambuStudio #4361: Constant Bambu Studio 1.9.3.50 crashes when slicing

Innova3DPrints, 2024-06-23:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro 22H2

### Additional system information

CPU: Intel(R) Core(TM) i7-3770 CPU @ 3.40GHz   
RAM: 32 GB
System: 64 bit
Graphics Card: Nvidia Quadro P1000

### Printer

X1

### How to reproduce

Select Supports, Tree Auto, Tree Slim
Slice
Crash
Need to reboot PC to have one clean slice before starting to crash.

### Actual results

Bambu Studio crashes.

### Expected results

Work great as previous BS version.

### Project file & Debug log uploads

[Pickleball Scorekeeper.zip](https://github.com/user-attachments/files/15944202/Pickleball.Scorekeeper.zip)
[log.zip](https://github.com/user-attachments/files/15944204/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

docshiv, 2024-06-23:
I can't get it to stop crashing even after a restart. Bambu Studio is unusable for me right now.

QingZhangBambu, 2024-06-24:
https://github.com/bambulab/BambuStudio/issues/3318
Hi, i could not reproduce it, could you try these methods? sometimes the CPU may not be stable enough to support multithreading.

Meagleson88, 2024-07-10:
I also keep having this when using tree supports (Auto) trying to use PETG as supports on PLA prints.

instantly crashes every time at 30% slicer/fed to printer! (Sending via wifi/cloud)