bambulab/BambuStudio #4364: Bambu Studio shuts off after starting the inner chamber camera on X1C

nymphet-tech, 2024-06-24:
### Bambu Studio Version

19.3.50

### Where is the application from?

Built from source by yourself or others

### OS version

Windows 11

### Additional system information

CPU: Intel(R) Core(TM) Ultra 9 185H   2.30 GHz
Memory: 32.0 GB (31.6 GB usable)
System type: 64-bit operating system, x64-based processor
Display card: NVIDIA® GeForce RTX™ 4050 Graphics, 6GB

### Printer

Bambu Lab X1C

### How to reproduce

1. Start Bambu Studio
2. Choose Device tab
3. Click on Play for the inner chamber Camera
4. Watch it tries to connect while displaying "Loading"
5. Bambu Studio quits

### Actual results

The app closes itself. Can't screenshot it. I can do a video. 

### Expected results

it should show me the video display of the inner printer chamber, 

https://github.com/bambulab/BambuStudio/assets/173640334/0de1ebda-fc0a-4da4-9c53-2c940665e486

### Project file & Debug log uploads

It isn't print profile related

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2024-06-24:
Please follow the wiki and config GPU for BambuStudio.
https://wiki.bambulab.com/en/software/bambu-studio/faq/live-view

nymphet-tech, 2024-06-24:
I closed all Bambu remote access to the camera.
Used the registry file as was explained in this wiki and it is added to the
registry.
Windows Media Player is installed and working properly.
I do not receive any error messages. The software quits without any
notifications.
The device camera connects to the old laptop, cell phone and the tablet
with no issues. Not to the brand new laptop.

None of the items in the wiki seems to apply to my case.

Regards,
Nilay Engin
505-440-2172
https://www.linkedin.com/in/NilayEngin/

nymphet-tech, 2024-06-24:
I  reinstalled the app. The problem persists.
I uninstalled and reinstalled the app. Problem persists.
Bambu Studio has no issue connecting to the live printer and my printer is
online. I was able to send the last printer from this device with no
problems. I just can't see the video.

Regards,
Nilay Engin
505-440-2172
https://www.linkedin.com/in/NilayEngin/

nymphet-tech, 2024-06-24:
Changed it in NVIDIA Control Panel.
Now the program is crashing instead of quietly quitting.

Also changed it in Wondiws graphics display.

Nome of them is working.

bambu123, 2024-06-25:
Unfortunately, it seems that your environment is no longer compatible with the old version of WMP. We need to wait until we refactor the player to solve it.

nymphet-tech, 2024-06-25:
OK, but this ticket needs to stay open till then.

I will try another slicer like ORca because friends tell me the camra works
with it.
I just bought A1 mini. Will it be the same problem?

Regards,
Nilay Engin
505-440-2172
https://www.linkedin.com/in/NilayEngin/