bambulab/BambuStudio #4365: Bambu Studio crashing when clicking "Preview" tab after cloning object

rrowhe4d, 2024-06-24:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro - Version: 10.0.22631

### Additional system information

CPU: i9-13900K 
RAM: 64GB / 6400 MHz / DDR5
GPU: Nvidia RTX 4090

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1) Right click on initial object to be printed and click "clone"
2) Select number of clones you want, ex.: 59 (total = 60)
3) Click "Preview" tab
4) Wait for crash as it loads all of the different objects to preview

### Actual results

Single object preview works fine, only happens when I try to clone many of the same objects and click preview tab. Tried it with different clone quantities like: 9, 59, 69, etc. It seems to work fine when selecting small clone quantities like or 2. It fails when selecting clone quantities above 2.

### Expected results

Bambu Studio just closes itself and needs to be restarted. Sometimes it gives an error message at the following startup to load the project before the crash.

### Project file & Debug log uploads

[Bambu Crash.zip](https://github.com/user-attachments/files/15949973/Bambu.Crash.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

rrowhe4d, 2024-06-24:
User @QingZhangBambu linked to [this](https://github.com/bambulab/BambuStudio/issues/3318) solution which seems like a feasible workaround to the problem.

closed 2024-06-24 by rrowhe4d

evlas1013, 2025-02-11:
Having this issue any time the Preview tab is opened. Even if no model is loaded. Adjusting Affinity did nothing.

i7-7700, Windows10

v1.10.1.50