bambulab/BambuStudio #4369: K-Factor get not applied

nmkr, 2024-06-24:
### Bambu Studio Version

1.9+

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambulab A1

### How to reproduce

Use 1.8.4.51 with A1 Firmware 1.3.0 
use 1.9+ with A1 Firmware 1.3.0 

### Actual results

Use 1.8.4.51 with A1 Firmware 1.3.0 
Profiles not implemented yet in the old Studio, set K-Factor manual to 0.020-0.025 depend on Material
Result is fine with no overextrusion and sharp edges and round corners, all good

use 1.9+ with A1 Firmware 1.3.0 
Profiles implemented, created with the same K-Factor as set in old Studio (0.020-0.025) 
No matter what k-factor is set the end result is the same with severe overextrusion and round edges and over round edges.

Tried even funky Factors from 0.000 to 0.0400 with the same results all the time.
as soon i slice it in the old studio it came out fine again.

seams a combination of Firmware and Studio because on Firmware 1.2.xx i was already on 1.9.1X Studio without issues (no profiles, manual kfactor) 

### Expected results

Same Print Quality result with Custom Profiles that not ignore the K-Factor value

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2024-06-27:
Can you use Studio 1.9+ and firmware 1.3.0 to perform slicing and printing again with a simple model, and share your Studio logs and printer logs?

nmkr, 2024-06-27:
hello, can do it over the weekend
do you have any hint beforehand what can cause it maybe? so i maybe can test that too??

attached also an image where the problem is visible
above 1.8.X manual factor
below 1.9.X with profile

above clear text, sharp edges and black top layer no overextrusion on edges
below bloated text, rounded edges and overextrusion on the top

[image]

zhimin-zeng-bambulab, 2024-06-27:
It's OK.
1. On the device page, make sure your ams has set up the corresponding PA profile.
2. Send a print of a simple model to verify the print quality
3. Share your Studio and printer logs

nmkr, 2024-07-01:
Hello, 

Had time to create the test, have a look at the attached logs and project files.

First print was with 1.8.X and second with 1.9.X
K-Factors like shown above set to the same value as 1.8.X

[log.zip](https://github.com/user-attachments/files/16051233/log.zip)
[projects.zip](https://github.com/user-attachments/files/16051236/projects.zip)

[image]

hope this helps and is correct so far, thanks in advance

nmkr, 2024-07-12:
@zhimin-zeng-bambulab Are the log files okay?

zhimin-zeng-bambulab, 2024-07-15:
Sorry for the late recovery. According to the investigation, it may be a problem with the printer. Could you provide the printer log? There are only Studio logs.

nmkr, 2024-07-15:
hello why should it a problem with the printer? its an error from the slicer because with the old version it works as expected.
how can i extract the printer log files?

zhimin-zeng-bambulab, 2024-07-18:
https://wiki.bambulab.com/en/a1-mini/troubleshooting/how-to-upload-log

You can refer to the wiki above

closed 2024-08-09 by nmkr