bambulab/BambuStudio #4370: Keypresses not being recofnized.

mihnt81, 2024-06-24:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab github releases

### OS version

Linux Mint 21.3

### Additional system information

CPU: Ryzen 5 3600
RAM: 16GB
GPU: 6650XT 8GB

### Printer

P1S

### How to reproduce

1. Select model on plate
2. Use keybind
3. Sometimes nothing, sometimes it works

### Actual results

Sometimes the key presses work. For example I was trying to reprint the failed parts of the AMS drybox stuff. When I was trying to delete and rotate some of them, neither backspace, delete, or R would do anything to the models. However, on some of them, they would respond accordingly.

### Expected results

If I press R, go into rotate mode. If I press delete or backspace, the model is removed.

### Project file & Debug log uploads

Project File: This wasn't specific to just one file. It was doing it on all of them.

Log File: 
[log.zip](https://github.com/user-attachments/files/15960449/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

caco3, 2024-06-27:
I can confirm that, under Linux, sometimes a key press (eg. to slice) does not work.

LiZ-Li-BBL, 2024-07-05:
Thank you for reporting the problem, we will check this issue.