bambulab/BambuStudio #4371: Alternate nozzle sizes not supported

davidjdixon, 2024-06-24:
### Bambu Studio Version

1.9.3

### Where is the application from?

Bambu Lab github releases

### OS version

14.5

### Printer

Bambu X1 Carbon

### How to reproduce

1. Create a printer profile with a nozzle size not provided by Bambu Lab (eg 1mm, as supported by the Panda Revo 3rd party hotend)
2. Send any print to the printer and watch the error

### Actual results

An error screen is presented about nozzle size mismatch with the configured size on the printer with no option to bypass.

### Expected results

Previous versions provided a warning, but allowed a user to continue.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [X] Log file
- [ ] Project file

davidjdixon, 2024-07-26:
@walterwongbbl is there any chance this recent change can be reverted? I use custom nozzle sizes with my X1C and this is a real blocker for me using the latest version of Studio.

walterwongbbl, 2024-07-31:
@davidjdixon 
I will forward your issue to the product department.

davidjdixon, 2024-07-31:
> @davidjdixon 
> 
> I will forward your issue to the product department.

Thanks. Does that mean this change was purposeful?

walterwongbbl, 2024-07-31:
@davidjdixon 
[image]
Hi， You can set the nozzle type and nozzle diameter in the Printer Parts section of the device page to match the printer. Then try to see if it will still be blocked.

davidjdixon, 2024-08-06:
@walterwongbbl i think you may have misunderstood the issue. It is not that the nozzle size cannot be set, but it cannot be set to a value other than 0.2, 0.4, 0.6 and 0.8, so doesnt allow for custom nozzle sizes for 3rd party hotends, such as those possible via the Panda Revo.

davidjdixon, 2025-01-18:
This has still not been updated and it prevent any customer nozzle sizes being used.

davidjdixon, 2025-01-20:
@walterwongbbl are you able to comment on this?