bambulab/BambuStudio #4372: Do a prime line for each color change when printing object by object

VFLashM, 2024-06-24:
**Is your feature request related to a problem? Please describe.**

I often print assemblies of different colors on the same plate in object-by-object mode.
As in one assembly, color change, another assembly, color change, one more assembly and so on.

For the first color printer does a priming line, which helps to normalize the flow of the filament.
Unfortunately, it doesn't do that after any of the other color changes.
Not priming the flow results in an over-extruded blob of plastic at the point where printer begins to extrude the assembly.

**Describe the solution you'd like**

Add an option to do a priming line each time when filament changes between objects when printing object-by-object.
Or simply do it every time when filament changes while extruder is at level 1 (and no prime tower exist).

**Describe alternatives you've considered**

Prime tower could help, but right now it cannot be enabled when printing object-by-object.
Most likely because each object would require a separate tower.
Alternative is to make it possible to use a prime tower(s) when printing object-by-object.
Prime tower can be even be a single-layer one.

SaltWei, 2024-06-26:
Skirt generation has already considered this
[image]

VFLashM, 2024-06-27:
Skirt is a good idea. Unfortunately, it doesn't work that well.

There are also some bugs with it:
1. Skirt of the first object encircles everything, not only the first object.
2. Skirts of the objects except first ignore skirt loop count (always one loop).
3. Most importantly, the skirt of the second (and third) object is much, much closer to the object for some reason. This is clearly visible on the screenshot @SaltWei attached. The blob at the beginning of the extrusion causes skirt to melt with the object, ruining the finish.