bambulab/BambuStudio #4375: X1C print profiles for Prusament filaments - where to find, or how to make?

3dafterlife, 2024-06-25:
**Is your feature request related to a problem? Please describe.**
The generic profiles are not the best for printing Prusament filaments

**Describe the solution you'd like**
X1C print profiles for Prusament filaments - where to find, or how to make?

**Describe alternatives you've considered**
just use generic profiles, but ...

**Additional context**
Add any other context or screenshots about the feature request here.

darth-liam, 2024-06-26:
On the "Prepare" screen in BambuStudio, click the Gear icon (rightmost icon in the sidebar) on the "Filament" dropdown. You can select pre-existing filament profiles under "System Filaments" or create your own under the "Custom Filaments" option by clicking "Create New". You will be asked to provide Vendor, Type, and Serial information for the filament, and you can then choose if you would like to create the filament preset based on your current filament.

You can improve print profiles for filament by running Flow Dynamics and Flow Rate Calibration tests under the "Calibration" tab in BambuStudio. Enter your nozzle and plate information for the printer, then choose the filament you would like to test. These two separate tests will gradually improve print quality for the examined filament until finding the optimal settings. After finishing the calibration tests and clicking "Finish", you will then be given the chance to save the optimized settings to your print profile. I recommend running these tests with the filament as close to your typical printing conditions as possible (dried filament, usual ventilation, door and lid open or closed, etc.). Hope this helps!

3dafterlife, 2024-06-28:
Thank you!Very useful.   Esun publishes profiles for bl printers  -filament and process - for their filaments.  Especially useful for HS filaments. I was hoping for the same for Prussia filaments….Thank you again.