bambulab/BambuStudio #4377: No files on SD card in "device"

Folxam, 2024-06-26:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab A1

### How to reproduce

Go to Device page and click SD card.

### Actual results

 The page says "File does not exist. [10]" but there are files on the card, shows locally on printer.
[image]

### Expected results

I should see my files.

### Project file & Debug log uploads

[Bambu+Spool+Voroni+V1+Profile_Voronoi Spool Narrow.gcode.zip](https://github.com/user-attachments/files/15982902/Bambu%2BSpool%2BVoroni%2BV1%2BProfile_Voronoi.Spool.Narrow.gcode.zip)
[debug_Wed_Jun_26_06_37_40_17248.log.zip](https://github.com/user-attachments/files/15982928/debug_Wed_Jun_26_06_37_40_17248.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2024-06-26:
Is there any file in the Model tab?

Folxam, 2024-06-26:
Yes, of course. Now I feels stupid :)
(It could be a bit more clear that you're in a tab called Timelapse though)

I think I clicked everything including the Timelapse and Model,
but now I'm not so sure :)

Thanks!
Mvh Jonas