bambulab/BambuStudio #4380: Bambu Matte PLA not printing

ImolaS3, 2024-06-26:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sonoma

### Additional system information

Realy not relevant

### Printer

X1 carbon with AMS

### How to reproduce

Try to print with bambu Matte blue PLA - the model is from here and I am printing the 'spirals'
https://www.printables.com/model/414209-botanical-portal-desk-toy/files

### Actual results

First off, i have 10+ years of 3D printing experience both using and designing machines as well as modifying and using off the shelf units. 
I spent a lot fo money on two X1C printers as I want to just print with them, not modify, fiddle, change, or anything else.

The print quality using Bambu Matte blue PLA directly out of the box is absolutely appalling. I am using the built-in settings for Bambu Matte, and even when I slowed every speed down by 50%, it was equally as bad. The top 30mm of the print is just a mess and the entire length of the section where it actually printed something like the spiral profile, it has voids and is just appalling.
[image]
[image]of money on two X1C printers because I want to just print with them, not modify, fiddle, change, or do anything else with them

### Expected results

I expect a decent-quality print, not 3 attempts that have all gone directly into the bin.
I also expect bambu to
A) create working profiles for their filaments
B) sell filament that is ready to use out of the box (I suspect moisture might be part of the issue as well as a the poor profile)

### Project file & Debug log uploads

Not relevant, 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ImolaS3, 2024-06-26:
I should also mention that using a standard PLA with standard profile, these issues do not exist, it is specific to the matte filament and/or profile

ttt888zzz, 2024-06-27:
Dear [ImolaS3](https://github.com/ImolaS3)
could you provide the original model？thanks

QingZhangBambu, 2024-06-27:
i will check this.

QingZhangBambu, 2024-06-27:
it seems that it does not stick to the print plate well.  I print one with default parameters, and it seems good.  if you do not use the default params, could you upload your 3mf files? and if it always fails, could you upload the machine log. 
[image]

nichi0626, 2025-01-09:
Has this issue been resolved. I'm having the exact same issue