bambulab/BambuStudio #4381: display stays black camera does not work sd card not recognize. i try to format an change cards but nothing

Uldits, 2024-06-26:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

win 11

### Printer

p1s

### How to reproduce

display stays black camera does not work sd card not recognize. i try to format an change cards but nothing

### Actual results

display stays black camera does not work sd card not recognize. i try to format an change cards but nothing

### Expected results

display stays black camera does not work sd card not recognize. i try to format an change cards but nothing

### Project file & Debug log uploads

display stays black
[01P09A440900089_06-26_10_01_45.690_v01.06.00.00_idx_1.log](https://github.com/user-attachments/files/15986497/01P09A440900089_06-26_10_01_45.690_v01.06.00.00_idx_1.log)
 camera does not work sd card not recognize. i try to format an change cards but nothing

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2024-06-27:
It seems to be a problem with the printer, please help create a ticket.

https://wiki.bambulab.com/en/p1/troubleshooting/how-to-upload-log

bambu123, 2024-06-27:
Regarding the issue of camera or liveview not working, please help send a studio log.
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

Uldits, 2024-06-27:
[crash_Fri_Jun_21_08_51_16_0_done.log](https://github.com/user-attachments/files/16009307/crash_Fri_Jun_21_08_51_16_0_done.log)
Hello there is a log file

gaolegao-lx, 2024-06-28:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

Uldits, 2024-06-30:
Problem are solved by bambu lab. It was someting with ssl certificates, but enyway i hawe to disconect printer from network to avoid such kind a problems.