bambulab/BambuStudio #4383: Modifiers Disappearing on Painted Parts

Shuwy05, 2024-06-26:
### Bambu Studio Version

v01.09.03.5

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1 Carbon

### How to reproduce

I have Text modifiers on a part that show up just fine. When I add paint to a part the modifier is on, which is a different color than the modifiers, the text modifiers will disappear. 

### Actual results

[image]
[image]
[image]
[image]

### Expected results

Should show the text in the specified color. 

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-06-27:
could you provide a gif or video , we will check this

Shuwy05, 2024-06-27:
> @Shuwy05 Could you share your 3mf file?

Absolutely, here you go! 
[Capra Roof .zip](https://github.com/user-attachments/files/16012038/Capra.Roof.zip)