bambulab/BambuStudio #4384: Filaments in AMS Not Accurate or Updating unless switching to another printer and back

ParaDiddle777, 2024-06-26:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Printer

All

### How to reproduce

1. Have multiple printers registered in Bambu Studio. I have 3 X1C, 1 P1P, 1 A1.  
2. When in the Prepare Tab if I click the filament dropdown, it used to show all the filaments in the printer's AMS as choices, but now that's inaccurate unless I switch to a new printer, and then back to the printer I want to use. 
3. There's a workaround but it used to work better and be accurate without switching back and forth.
4. This bug was introduced 2 versions ago BTW. I was hoping it would be fixed, but it hasn't.

### Actual results

In the above steps it shows the wrong filaments.

### Expected results

BS should show the filaments accurately.

### Project file & Debug log uploads

This happens on all projects. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

ttt888zzz, 2024-06-27:
could you provide a gif or video , we will check this

ParaDiddle777, 2024-06-27:
Sure: https://www.youtube.com/watch?v=5RY26QpBors 

Open BS, the printer's AMS filaments won't show. If you switch to another printer and back, they will show. Not a showstopper and there's a workaround but it's annoying when I try to open a project and choose my colors.

DanBao-Bambu, 2024-07-08:
[image]
You can try resync your material information from AMS.

ParaDiddle777, 2024-07-08:
It used to work without this. This solutoin is not acceptable as it changes the colors on the model and then it takes even longer to remap the colors. I don't want to sync all the filaments sometimes, just be able to pick and choose what I want from the AMS. Guaranteed you broke something 3 versions ago in this regard.

ParaDiddle777, 2024-07-08:
Also this solution still doesn't work. I have PETG loaded in my AMS and when I resync all, it puts totally generic filaments into the software, so this is NOT a workaround and completely broken anyways. I'm shocked you haven't tried to reproduct my steps, instead offering a solution that you clearly haven't tested.

DanBao-Bambu, 2024-07-09:
I'm sorry I misunderstood your question, and thank you for your prompt response. It made me realize my mistake, and I will analyze the issue.

ParaDiddle777, 2024-10-07:
This issue hasn't been fixed and it's highly annoying. This needs to work like it used to. It's the easiest way to assign filament types without messing up the colors. An AMS sync here is not useful, as it messed up the painting. The filament dropdown needs to automatically show loaded AMS filaments as the first/top choices. Please fix this.