bambulab/BambuStudio #4387: Setting wall generator to Arachne creates overhangs on certain small details. (Inconsistency in relation to layer height)

Nick420thStreet, 2024-06-27:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: AMD Ryzen 5 5600X 6-Core, Memory: 32 GB RAM, GPU: NVIDIA GeForce RTX 3060

### Printer

Bambu Lab A1 + AMS lite

### How to reproduce

1: Select "Arachne" as wall generator
2: Slice the model

### Actual results

When I use "Arachne" as the wall generator, the detection of thin walls is inconsistent in relation to layer height. So for bottom and top layers it will detect more of the thin wall than for the layers in between. This creates a 90 degree overhang for the top layers. The result when using the "classic" wall generator is as expected though. It is visible in the sliced preview, but I haven't tried printing it, since I don't want to waste filament on an ugly print. So it might just be a visual bug in the slicer app. There is no overhang warning when slicing either, which I do get when there is an actual overhang in my models and I haven't enabled support.

Screenshot: [image]

### Expected results

When referring to the 3d model that is being sliced, the "arachne" wall generator should produce the same wall length/thickness across all layers.

### Project file & Debug log uploads

[Keychain Dragon Strong Extruded v1.zip](https://github.com/user-attachments/files/16012978/Keychain.Dragon.Strong.Extruded.v1.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

MergingMonkey, 2024-12-27:
Hi, I can confirm and want to add another test project for analysis purposes:
[Test Part.zip](https://github.com/user-attachments/files/18260806/Test.Part.zip)

It contains three project process profiles: "Classic", "Arachne default" and "Arachne modified" (where I tried to get it fixed by modifying the arachne settings).

Here are the differences. Classic:
[image]

Arachne default, creating a seam at default settings:
[image]

Arachne with modified settings:
[image]

This is the part printed with Arachne default, where you can see the problem in reality:
[image]

dmitrii-galantsev, 2025-01-15:
@MergingMonkey Seeing the same issue in latest OrcaSlicer.
Lowering "Wall transitioning threshold angle" seems to help.

Below examples use your Test Part.zip.

<details>

OrcaSlicer 2.3.0-dev build 2ea2ab0 Appimage on Linux.

## 0. Classic:
[image]

## 1. Arachne - No change:
[image]

## 2. Arachne - 3 degrees:
[image]

## 3. Arachne - 3 degrees + 25% min wall width:
[image]

</details>

dmitrii-galantsev, 2025-01-15:
lowering "Wall transition length" and keeping "Wall transitioning threshold angle" as default helped too.

XunZhangBambu, 2025-01-21:
@NickOnTheAsteroid  Sorry for the delayed response . The method proposed by @dmitrii-galantsev  can generate such small features.

Wall transition threshold angle : a larger angle means a larger area where the variable line width is applied.
Wall transition length: A longer transition area length can reduce the bumps generated on the surface.
Minimum feature size: areas smaller than this size will be ignored.
Minimum wall width：The generated minimum line width. That means lines which are greater than the minimum feature size but smaller than normal width, will ultimately be generated to the minimum line width