bambulab/BambuStudio #4389: AMS-System löscht permanent alle Filament-Einstellungen

Perseusix, 2024-06-27:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

Ich habe leider ein Problem mit einem AMS-System.
Genauer gesagt speichert das AMS nicht mehr die Filament-Daten der einzelnen Slots. Spätestens beim Drucken sind dann die meisten Werte wieder gelöscht.

Ich habe ein Video für euch gemacht.

https://www.loom.com/share/855efd60972840b8b47b9d0958d1f3c1?sid=e0d211d1-bcdf-468c-8715-fe689ef38a0f

Hier ist zu sehen wie ich den Drucker auswähle und dann alle 4 Slots mit Filamenten belege. Man sieht auch wie die Slots dann wieder gelöscht werden vom Drucker oder AMS. Keine Werte bleiben erhalten.

Ich habe das auch mit Generic-Profilen ausprobiert, auch hier ist es gleich.

Das AMS kann ich so leider nicht mehr benutzen.

Was kann ich machen?

### Actual results

Ich habe ein Video erstellt.
Da kann man sehen wie sich ständig die Filamente löschen.

https://www.loom.com/share/855efd60972840b8b47b9d0958d1f3c1?sid=e0d211d1-bcdf-468c-8715-fe689ef38a0f

### Expected results

Verstehe diese Frage nicht.

### Project file & Debug log uploads

[Skadis Schale - Bambu-Fehler.zip](https://github.com/user-attachments/files/16019501/Skadis.Schale.-.Bambu-Fehler.zip)
[log.zip](https://github.com/user-attachments/files/16019502/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-07-15:
Could you enable trace service level and reproduce this problem again? You can modify the configuration file as shown below.
[image]
[image]

Then could you upload Studio logs and printer logs?

DucatiMNSTR, 2024-08-21:
Same problem here.