bambulab/BambuStudio #4390: Internal Bridge infill Fan On & Gaps on Internal Bridge Infill.  Need Expansion settings and fan off for Internal Bridge layer

kpdcompany, 2024-06-27:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu A1

### How to reproduce

Turn on Fan for bridge
Go to internal bridge infill layer on model(First "solid" layer of top solid layers)
Fan is on for that layer which causes warping and shrinking a that layer due to the fan.
We still want our Bridge fan at 80% but under no circumstances should this be FORCED for the first solid "bridge" layer.  We have had multiple part shrinkage issues due to this exact problem.  We would like to be able to adjust the fan for the bridge infill or just turn it off.  
"Fan on/off for Internal Bridge Layer"

[image]

For the same layer please see the gaps left, we would like an expansion setting for this layer or something so that it is not making these areas for print defects.  Weird to me is the hole, it has a "cube" centered on it but the solid layers deviate and make them closer to the hole on one side than the other, seems like something is up with this layer algorithm.  I think there used to be some sort of infill layer expansion or something to adjust this.

[image]
[image]

### Actual results

Fan is on during first solid top "bridge" layer.

### Expected results

Fan should not be forced on for solid bridge layer jsut because you have your bridge fan on.  Multiple parts require bridging over support which requires fan, we can't turn off the bridge fan to stop solid layer shrinking since we still have a REAL bridge to do.  This setting should not ever effect the solid layer fans.  The first solid layer should not be a "Bridge".  This is what is causing problems.3

### Project file & Debug log uploads

[BRIDGE FAN ON SOLID ISSUE - PROJECT.zip](https://github.com/user-attachments/files/16019599/BRIDGE.FAN.ON.SOLID.ISSUE.-.PROJECT.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

kpdcompany, 2024-06-27:
another fix for the time being would be just a way to use modifiers for fan settings.

QingZhangBambu, 2024-07-04:
the bridge area had been fixed. and will release in the next version
about the fan control, i will test it.

yzheka, 2025-06-04:
I would love to have ability to control fan speed for internal/external bridges like it is in Orca.