bambulab/BambuStudio #4391: ubunut 24.04 version of Bambu studio won't slice.

hainjedaf, 2024-06-27:
### Bambu Studio Version

1.9.3.50 ubuntu 24.04 appimage

### Where is the application from?

Bambu Lab github releases

### OS version

Debian 12 Sid / Unstable

### Additional system information

Ryzen 9 3900x 64Gb memory, 512Gb NVMe

### Printer

X1Carbon

### How to reproduce

-load a stl or 3mf file
-select slice plate
Error occurs:
boost::filesystem::create_directory: No such file or directory [system:2]:
"/tmp/bamboo_model/Thu_Jun_27/21_12_35#109292#47/Metadata"

### Actual results

Bambu Studio Locks up.

### Expected results

Sliced plate, option to send to printer

### Project file & Debug log uploads

[bambu-lab-filament-spool-winder-upgraded-model_files.zip](https://github.com/user-attachments/files/16020304/bambu-lab-filament-spool-winder-upgraded-model_files.zip)
[debug_Thu_Jun_27_21_logs.zip](https://github.com/user-attachments/files/16020306/debug_Thu_Jun_27_21_logs.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

lanewei120, 2024-06-28:
could you remove this folder or change the folder's permission and try again?
/tmp/bamboo_model/Thu_Jun_27/21_12_35#109292#47/Metadata

it is likely this folder has been occupied by other account which not allow bambustudio to create/write

hainjedaf, 2024-06-28:
I did reboot between attempts. /tmp is a ramdisk so should be empty after reboot.

Yes it now works.

closed 2024-06-28 by hainjedaf