bambulab/BambuStudio #4398: Warning Message Prevents Printing CF Material

max-sun-hongyi, 2024-06-28:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

MacOS 14.3.1 Sonoma

### Additional system information

MacBook Pro M1 Pro 

### Printer

Bambu Lab P1S with AMS

### How to reproduce

**To reproduce this issue, print any model with PLA-CF material in Bambu Studio 1.9.3.50. with hardened steel hotend installed** 

I am printing with PLA-CF filament, with harderned steel hotend, I used to be able to print such filament but since I updated the Bambu Studio to the latest version (1.9.3.50) I get the below warning message asking me to rectify the hotend issue (printing cf material may damage hotend) and I am unable to send file to printer. 

I followed this wiki pages (https://wiki.bambulab.com/en/x1/maintenance/replace-complete-hotend) to change the hotend to hardened steel and changed the hotend type in the software to 'hardened steel" 

Temporary solotion is to use a previous version of the Bambu Studio (1.8.4.51) In which I can print with such filament. 

My suspision is that a warning feature was intruduced to the software to prevent printing cf material with stainless hotend but the logic failed to properly check the type of hetend  

[image]

### Actual results

Unable to send gcode to printer due to this warning message
[image]

### Expected results

It is expected that the gcode will be sent to printer to begin printing process 

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16029505/log.zip)
[test block.3mf.zip](https://github.com/user-attachments/files/16029519/test.block.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2024-07-09:
Hi. @max-sun-hongyi 
You can go to the device page and set the nozzle type, which can solve your problem.

Phil-Barker, 2025-04-09:
I get the same warning though can now click continue to get past it

However I have a P1S that I've put the hardened steel nozzle on and there's no option to set that in printer settings

[image]

loihuynh97, 2025-04-24:
[image]
Just config look like image