bambulab/BambuStudio #4399: 从solidworks导出包含中文部件名称的step文件导入后会显示"step文件中的部件名称包含非UTF8格式的字符！"

weijinzhongyi, 2024-06-28:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

win 11

### Printer

A1

### How to reproduce

导入从solidworks导出的包含中文的step文件(文件名和部件名), 都会出发

### Actual results

[image]

但软件内都能正确识别中文的部件名称

其他的建模软件不会报这个错, 如onshape.

### Expected results

优化一下step的解释器

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16032719/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

MackBambu, 2024-07-01:
可以分享一下step文件吗？

swmp, 2024-07-08:
我也有这个问题
[md_gfx.zip](https://github.com/user-attachments/files/16131499/md_gfx.zip)

MackBambu, 2024-07-11:
这只是一个提示，不会影响文件的导入流程。
多数CAD软件在保存文件时，中文字符会采用GBK编码，bambu studio在读取时会尝试使用GBK读取文件名和装配名，但是仍然有可能是存在其他编码的字符导致读取文件名或零件名乱码。这并不能分辨，因此仅仅弹出对话框做出提示。

closed 2024-07-15 by MackBambu

junchao98, 2025-03-20:
这个提升能关掉吗，导入很多零件时，每个零件都要弹一次

jiangxu163email, 2025-03-22:
该提示严重影响使用体验，希望尽快修复