bambulab/BambuStudio #4405: Bambustudio network test failed - no Route to Host

Knaudler, 2024-06-29:
### Bambu Studio Version

01.09.03.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10 recent

### Printer

Bambu A1 mini

### How to reproduce

Bambustudio
network test
Test Storage and test storage upgrade

### Actual results

Sat Jun 29 20:12:10:test storage start...
Sat Jun 29 20:12:10:[test_oss]: url=http://upload-file.bambulab.com
Sat Jun 29 20:12:10:test storage failed
Sat Jun 29 20:12:10:status=404, body={"error_msg":"404 Route Not Found"}
, error=

---

Sat Jun 29 20:12:29:test storage upgrade start...
Sat Jun 29 20:12:29:[test_oss_upgrade]: url=http://upgrade-file.bambulab.com
Sat Jun 29 20:12:29:test storage upgrade failed
Sat Jun 29 20:12:29:status=404, body={"error_msg":"404 Route Not Found"}
, error=

### Expected results

I think it should be ok 

### Project file & Debug log uploads

Sat Jun 29 20:13:56:test bambulab start...
Sat Jun 29 20:13:56:[test_bambulab]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?name=slicer&version=01.09.03.50&guide_version=00.00.00.01
Sat Jun 29 20:13:56:test bing start...
Sat Jun 29 20:13:56:[test_bing]: url=http://www.bing.com/
Sat Jun 29 20:13:56:test http start...
Sat Jun 29 20:13:56:test storage start...
Sat Jun 29 20:13:56:[test_oss]: url=http://upload-file.bambulab.com
Sat Jun 29 20:13:56:test storage upgrade start...
Sat Jun 29 20:13:56:[test_oss_upgrade]: url=http://upgrade-file.bambulab.com
Sat Jun 29 20:13:56:test plugin download start...
Sat Jun 29 20:13:56:test storage download start...
Sat Jun 29 20:13:56:[test_oss_download]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?slicer/plugins/cloud=01.09.03.00
Sat Jun 29 20:13:56:test bing ip resolved = 2.20.142.187
Sat Jun 29 20:13:56:test bing ok
Sat Jun 29 20:13:56:test http ok
Sat Jun 29 20:13:56:test bambulab ip resolved = 172.64.152.215
Sat Jun 29 20:13:56:test bambulab ok
Sat Jun 29 20:13:56:test storage upgrade failed
Sat Jun 29 20:13:56:status=404, body={"error_msg":"404 Route Not Found"}
, error=
Sat Jun 29 20:13:56:[test_plugin_download]: downloadurl=https://public-cdn.bblmw.com/upgrade/studio/plugins/01.09.03.01/68969899e2/win_01.09.03.01.zip
Sat Jun 29 20:13:56:test storage failed
Sat Jun 29 20:13:56:status=404, body={"error_msg":"404 Route Not Found"}
, error=
Sat Jun 29 20:13:57:downloading 5%
Sat Jun 29 20:13:57:downloading 10%
Sat Jun 29 20:13:57:downloading 10%
Sat Jun 29 20:13:57:downloading 15%
Sat Jun 29 20:13:57:downloading 20%
Sat Jun 29 20:13:57:downloading 20%
Sat Jun 29 20:13:57:downloading 25%
Sat Jun 29 20:13:57:downloading 30%
Sat Jun 29 20:13:57:downloading 30%
Sat Jun 29 20:13:57:downloading 35%
Sat Jun 29 20:13:57:downloading 40%
Sat Jun 29 20:13:57:downloading 40%
Sat Jun 29 20:13:57:downloading 45%
Sat Jun 29 20:13:57:downloading 50%
Sat Jun 29 20:13:57:downloading 50%
Sat Jun 29 20:13:57:downloading 55%
Sat Jun 29 20:13:57:downloading 60%
Sat Jun 29 20:13:57:downloading 60%
Sat Jun 29 20:13:57:downloading 65%
Sat Jun 29 20:13:57:downloading 70%
Sat Jun 29 20:13:57:downloading 70%
Sat Jun 29 20:13:57:downloading 75%
Sat Jun 29 20:13:57:downloading 80%
Sat Jun 29 20:13:57:downloading 80%
Sat Jun 29 20:13:57:downloading 85%
Sat Jun 29 20:13:57:downloading 90%
Sat Jun 29 20:13:57:downloading 90%
Sat Jun 29 20:13:57:downloading 95%
Sat Jun 29 20:13:57:downloading 100%
Sat Jun 29 20:13:57:test storage download ok
Sat Jun 29 20:13:57:downloading 100%
Sat Jun 29 20:13:57:test plugin download ok

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2024-07-09:
HI @Knaudler 
What specific issue did you encounter? This network test will not affect your Studio functionality.

KwisatzJim, 2024-12-13:
I've encountered this as well when running the network test:

Fri Dec 13 17:47:07:test storage upgrade failed
Fri Dec 13 17:47:07:status=404, body={“error_msg”:”404 Route Not Found”}
, error=
Fri Dec 13 17:47:07:[test_plugin_download]: downloadurl=https://public-cdn.bblmw.com/upgrade/studio/plugins/01.10.01.01/8f52eddd78/mac_01.10.01.01.zip
Fri Dec 13 17:47:07:test bing ip resolved = 104.117.244.107
Fri Dec 13 17:47:07:test bing ok
Fri Dec 13 17:47:07:test storage failed
Fri Dec 13 17:47:07:status=404, body={“error_msg”:”404 Route Not Found”}
, error=

Andyman14159, 2025-03-07:
Hi. Same problem here. Multiple machines on my account and logged into bambu studio but they all run on the same local network. Send to printer worked fine for several months but now suddenly cannot send to printer. Sam exact problem as listed by OP all network tests pass except Test Storage Upload failed and test Storage Upgrade failed. Please list a solution to this problem help.