bambulab/BambuStudio #4406: Filament Sync only pulls filaments from first installed printer. No other printer.

deftoner, 2024-06-30:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10, latest patch

### Additional system information

Ryzen 7 5400 + Nvidia 3070 + 64Gb (lenovo legion 5 pro)

### Printer

X1C + P1S, AMS in each printer

### How to reproduce

From main Bambu screen, under "Prepare" if you select the second printer (P1S in my case), and under "Filament" you click sync, will bring only filament installed/set in X1C (the first printer installed) and not the P1S. 

[image]

[image]

### Actual results

It only brings filament list from the first installed printer, and not the selected printer

### Expected results

Should bring filaments from the AMS from the selected printer.

### Project file & Debug log uploads

No project file needed. this happens with or without project opened 
[log.zip](https://github.com/user-attachments/files/16043184/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

deftoner, 2024-06-30:
Adding info: 
When hitting "print plate" and the "Send print job to" pops up, the filament list for each printer shows as it should. But still when working with a project the list of filaments is only from the first installed printer ams. 

Also, if you got to "Device" tab, and select each devices (and will bring filaments of each devices correctly) the sync will work one time, then wont work again if you change printer from menu

deftoner, 2024-07-02:
Sorry, me again, just in case I did a video if that makes easy to explain the bug/issue

https://youtu.be/ic5klJW8RpM

ttt888zzz, 2024-07-04:
Please have a try：
[image]

MadMax389, 2024-07-06:
The slicer profile is not tied to the selected printer at all. You have to select the desired printer on the device screen in order to sync the correct AMS. It's been this way since always. I have submitted an issue to have the option to select which printer's AMS to use on the slicer screen so you don't have to go back to the device screen every time you switch printers.

deftoner, 2024-07-12:
> Please have a try： ![同步耗材-github](https://private-user-images.githubusercontent.com/169250342/345710947-fd92ea3a-c015-431a-8e8f-007ec4d83527.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA4MDk4MjEsIm5iZiI6MTcyMDgwOTUyMSwicGF0aCI6Ii8xNjkyNTAzNDIvMzQ1NzEwOTQ3LWZkOTJlYTNhLWMwMTUtNDMxYS04ZThmLTAwN2VjNGQ4MzUyNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcxMlQxODM4NDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01OWM4OTgyNWU0NmI5Mzg1MzJhOTVhNWYxOWJkMDgyNjhmNTcxNDFkNGI2NWNhM2M5NjNjYmYzYjM1ZWNlMTQ4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.P7k14y7Y3Q43iW7FUsbIhb5fEGCmiWUNFleHioQrMC4)

That way works. If I go to device each time and change the printer there and then I come back to prepare. Its definitely not user friendly or intuitive, because you have the printers right there in prepare, why need to go to another place to change the printer and then come back to sync the filaments, on a list, under a printer name, but is tied to another option in another tab, instead of the listed there? 

Is SO not intuitive that I felt that was a bug. 

May be a drop down to select the devices there in profile? or as @MadMax389 said, a way to select the AMS to sync from. 

I'm probably the issue, myself, since I work in QA for the past 10 years and I tried to make sense of the GUI, and I guess the big majority of users only have one printer. 

Another thing that I found annoying was to download a design that is tagged "X1C" but then the profile is A1. but  you can still sync your AMS, slice the plate,, send it to the printer. and is not until you go to the printer (physically) and select that design to print that will say that the profile is not intended for that printer. I think that a warning when you are sending a A1 profile to an X1C would be nice too but thats another subject

deftoner, 2024-07-12:
Apparently is not a bug, bus just a "non lineal or intuitive design" of the software that is that bad that looks like a bug :D

closed 2024-07-12 by deftoner