bambulab/BambuStudio #4409: Slicers says the print head retracts and goes to safe position but while printing it goes straight to prime tower knocking over support

Agroet, 2024-06-30:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu lab A1

### How to reproduce

Slice an object with overhang and enable tree support with AMS. The slicer says it retracts and moves to safe position (for example G1 X0 Y127.956 in image)

[image]

### Actual results

The print head goes straight to the prime tower, skipping retraction and thus Z-hop. It does this almost all layers, making a loud clicking noise. The move knocks over weak supports, forcing stronger supports. It is visible the print head does not Z-hop. The Z-hop happens on the prime tower. Meaning, it moves to the tower while not retracting or Z-hop. And then it hops, standing still.

https://github.com/bambulab/BambuStudio/assets/174253877/156dcf0b-0ff2-4b3d-978e-2234c38ef80b

### Expected results

The print head should Z hop as to not disturb the support.

### Project file & Debug log uploads

[Schuifoperatie debug 2.zip](https://github.com/user-attachments/files/16044760/Schuifoperatie.debug.2.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

gaolegao-lx, 2024-07-02:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

Agroet, 2024-08-13:
Disabling the “Reduce infill retraction” option has resolved the issue.