bambulab/BambuStudio #4410: Camera Issue

mnott, 2024-06-30:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 14.5

### Printer

X1C

### How to reproduce

Camera is giving bad results. I’ve very good lighting in the chamber (LED strips etc.), and the picture is normally fine; then suddenly this happened:

[image]

Following [this thread](https://forum.bambulab.com/t/camera-noise-pixelation/13071/19), switching back and forth between 720p and 1080p, the image appeared fine again…

[image]

Until, a very sort while later, JPG artifacts kick in:

[image]

Which get worse over time:

[image]

The same thing happens if I just view the picture from the Mobile app.

### Actual results

Bad picture (see above)

### Expected results

Has not happened ever before. Some JPG artefacts are ok, but not at this level, and certainly not the additional ISO noise.

### Project file & Debug log uploads

There's no log to attach for this. This is independent from the print file.

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2024-07-01:
Will other methods be effective without switching resolution on the printer: for example
1. Stop playing on Studio and start playing again
2. Restart Studio

ECKEROTH, 2024-07-03:
> Will other methods be effective without switching resolution on the printer: for example
> 
> 1. Stop playing on Studio and start playing again
> 2. Restart Studio

NO - that dont fix the issue

bambu123, 2024-07-04:
It is estimated to be a problem with the UVC camera. You can raise a ticket, and the camera will be replaced after the after-sales confirmation.
https://wiki.bambulab.com/en/x1/troubleshooting/upload-log