bambulab/BambuStudio #4411: Missing layers and faulty infill

NVNDO, 2024-06-30:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

P1S, A1M

### How to reproduce

Just slice the file and check the missing and erroneous layer

### Actual results

BambuStudio: 5 missing Layer
[image]

BambuStudio: Missing Infill
[image]

BambuStudio: Missing Infill
[image]

BambuStudio: Missing Infill
[image]

### Expected results

OrcaSlicer: How it should be!

[image]

[image]

[image]

[image]

### Project file & Debug log uploads

[Assembly.zip](https://github.com/user-attachments/files/16044825/Assembly.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2024-10-12:
@NVNDO it had fixed and will be released in the next version, please waiting.

QingZhangBambu, 2024-10-12:
[image]
![image](https://github.com/user-attachments/assets/63febba1-846c-4d0a-a720-9b863772a39

NVNDO, 2024-10-12:
> @NVNDO it had been fixed and will be released in the next version, please wait.

Oh hey! That's great! I did not encounter this problem with other parts, but will need to print similar ones again in the near future, so I assume we'll have it fixed by then. 

Thank you so much! 🙏🏼

QingZhangBambu, 2024-10-12:
@NVNDO OK. and if it still something wrong, you can upload 3mf files, we will check and fix

NVNDO, 2024-10-12:
I will do so if problems persist.