bambulab/BambuStudio #4412: Swapped Y & Z parameters in model scale adjustments

brain-slug2102, 2024-06-30:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 14.5

### Printer

Bambu Lab A1

### How to reproduce

1. Add a model to the build plate in Bambu Studio
2. Select "Scale" from the Top menu above the build late
3. A dialogue box appears with XYZ adjustments
4. Deselect "Uniform Scale"
5. Altering the Value for "Y" Adjusts the height of the model, which should be Z
6. Altering the value for "Z" adjusts the depth of the model, which should be Y
7. It appears that X and Y labels/functions have been swapped in "Scale"
8. Confirmed Z=height and Y=depth in "Move" settings

### Actual results

Screenshots show when Y is selected in the "Scale" menu, the numbers and arrows that demonstrate what is changing, all change the parameters of the Z axis, the other screenshot shows the opposite, when changing the values of Z, it actually shows the arrows and changes the parameters of Y axis

[image]
[image]

### Expected results

When adjusting the values of Y and Z, they should change the parameters of the corresponding axis

### Project file & Debug log uploads

[Y Z swap.3mf.zip](https://github.com/user-attachments/files/16045588/Y.Z.swap.3mf.zip)
[debug_Fri_Jun_28_08_03_04_1576.log.0.zip](https://github.com/user-attachments/files/16045590/debug_Fri_Jun_28_08_03_04_1576.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-07-02:
In the future, we will increase the scaling of the world coordinate system

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00