bambulab/BambuStudio #4414: A G-code path goes beyond the max print height

MichaelMcSw, 2024-06-30:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro version 23H2 OS build 22631.3737

### Additional system information

Processor	Intel(R) Core(TM) i7-10510U CPU @ 1.80GHz   2.30 GHz
Installed RAM	32.0 GB (31.8 GB usable)

System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

### Printer

Bambu Lab A1 0.6 nozzle

### How to reproduce

My model is 88mm tall. I am printing with Bambu PLA Aero  using 0.6 nozzle with the Spiral Vase mode. 
When I look at the post-slice layers, the top layer is numbered 25440 and the next layer down is 289.

If I set the nozzle as 0.8 or 0.4 it slices without problems, only with the 0.6 nozzle do I get the bogus top layer.

### Actual results

[image]

### Expected results

Shouldn't have bogus top layer

### Project file & Debug log uploads

[BT-20-.6mmNozzle-VaseMode.zip](https://github.com/user-attachments/files/16046642/BT-20-.6mmNozzle-VaseMode.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

skysyk, 2024-07-02:
I've encountered this issue under other slicing situations, I believe with ironing enabled for a print that has minimal surface area at the top of the model. Profile was 0.12mm fine with a 0.4mm nozzle. Unfortunately I do not have an example project to upload.

XunZhangBambu, 2024-07-12:
@MichaelMcSw 
Hello,you have set a large outer wall line width, but your model is very thin in the height range of 68.10 to 75.6, and it cannot accommodate the outer wall of the width you specified. You can check this by disabling  "sprial vase" The solution is to  set a smaller outer wall line width.
[image]
[image]

UBaer21, 2024-07-24:
Got that error with a tilted cylindrical part (small final layer area) - when deactivating "precise Z height" the error was gone

the height slider shows a wrong height for the last bit
[image]

but that number can't be found in the g-code.